MRSD2018/GroundsBot #29: Conduct highway department interview

davidjevans, 2017-09-07:
Interview at least 1 person from market to understand goals and problems of groundskeeping in market.

In person interview is preferable, but over the phone works too.

jphaneuf, 2017-09-25:
not enough personnel, can't add people, can hire temporary
focus is mainly on roads (paving, drainage, landscaping crew)
most mowing out on contract
spend ~ 1 million 
if want more mowing something has to give
not all of the municipalities , there's a misunderstanding of what the responsibilities are
in the burrows only responsible for certain parts, state is responsible
they do herbicide spraying

jphaneuf, 2017-09-25:
comes down to balancing funding
allocated reasonably well
changed the way they did mowing this year, changed contract, partial mowing and then a full mow

jphaneuf, 2017-09-25:
got a lot of complaints from that, was getting overgrown

jphaneuf, 2017-09-25:
why can't we ask workers to pick up trash as well?

jphaneuf, 2017-09-25:
would mow more just for aesthetics

jphaneuf, 2017-09-25:
if you don't mow trees start to grow

jphaneuf, 2017-09-25:
ramps/intersections there are sight distance issues

jphaneuf, 2017-09-25:
doesn't have a lot of equipment, most of it put out to bid

jphaneuf, 2017-09-25:
usually have a month to mow from a bid

jphaneuf, 2017-09-25:
-send email to jason to get contractor names

jphaneuf, 2017-09-25:
contract might say 'within 1 ft of trees' etc..., people go out to check specs

jphaneuf, 2017-09-25:
large tractors with pull behind mower decks

jphaneuf, 2017-09-25:
4-5 man crew on staff put in place in april w/ weedwackers, small tractor w/ mowing deck, go around and hit areas not on contract, large fields, react to complaints

jphaneuf, 2017-09-25:
any kind of job on the road is in a dangerous job, in traffic, mowing , not on slopes where tractor can flip, use weed whacker instead, for the most part less dangerous than patching potholes

jphaneuf, 2017-09-25:
automation: not yet, still low bid w/ contractors, was trying to find some equipment for picking up garbage, some state had some equipment to do that, cheapest most effective way is still people.  mowing is still in that area, not exploring that way.  don't want to limit a contractor's means or methods.  most creative contractors will win bid.  have seen some remote ones, but mostly just using big tractors.

jphaneuf, 2017-09-25:
there's a push / awareness about green space and storm control, reducing impermeable pavement area, the challenge is maintenance of it.  maintenance budget hasn't changed in several years, with this push for green infrastructure, giving water a place to absorb into ground ,  anything penndot can do to help is useful, several storm water retention places which require maintenance.  who's going to do it, where's the money coming from?

jphaneuf, 2017-09-25:
do a lot of beautification permits, folks paying for that and maintaining.  Tend to do just what they need to maintain safety,

jphaneuf, 2017-09-25:
As PennDot increases green infrastructure, where does money come from?

jphaneuf, 2017-09-25:
pittsburgh wet weather, 28 billion project to reduce runoff, FEMA? hasn't affected penndot yet

jphaneuf, 2017-09-25:
south hills developed 1900s no sight at all into retention of storm water

jphaneuf, 2017-09-25:
have to figure out a way to maintain that greenspace

jphaneuf, 2017-09-25:
jason zang: jzang@pa.gov

closed 2017-11-17 by jphaneuf